gatecat/prjoxide #1: Building nextpnr looking for database files?

davidcorrigan714, 2021-01-05:
I'm trying to build nextpnr against prjoxide on windows and hitting this error:

`
  Touching "x64\Release\chipdb-nexus-bbas\chipdb-n.1CF2954C.tlog\unsuccessfulbuild".
CustomBuild:
  "The build of 'C:\Users\User\Desktop\nextpnr\CMakeFiles\c064314d5248ea648493cfbc4f07cfae\chipdb-nexus-bbas.rule' depends on 'C:\USERS\USER\DESKTOP\NEXTPNR\NEXUS\CHIPDB\CHIPDB-LIFCL.BBA' which is produced by the build of 'C:\Users\User\Desktop\nextpnr\CMakeFiles\9264e4eece416fcd2a7c2b4cbd1744b1\chipdb-LIFCL.bba.rule'. The items cannot be built in parallel."
  Generating chipdb/chipdb-LIFCL.bba
  thread 'main' panicked at 'called `Option::unwrap()` on a `None` value', prjoxide\src\database.rs:481:62
  stack backtrace:
     0:           0x7bf046 - std::backtrace_rs::backtrace::trace_unsynchronized::h643d2cbff6704dbc
     1:           0x76a6e4 - <std::sys_common::backtrace::_print::DisplayBacktrace as core::fmt::Display>::fmt::hc5bfef9f5bc416d7
     2:           0x81ad9c - core::fmt::write::h1417336d9d869ecc
     3:           0x7a8e23 - std::io::Write::write_fmt::hda58e0f5d6826864
     4:           0x7a0e22 - std::panicking::default_hook::{{closure}}::h5a2888d9d53a4e94
     5:           0x7a0b32 - std::panicking::default_hook::hb32a684211ad1286
     6:           0x7a144c - std::panicking::rust_panic_with_hook::h404a72f576cde3a5
     7:           0x76abfa - std::panicking::begin_panic_handler::{{closure}}::h07abfc44f5bd25d7
     8:           0x76a84f - std::sys_common::backtrace::__rust_end_short_backtrace::hdd961d4b0da77bf7
     9:           0x7a1072 - rust_begin_unwind
    10:           0x81f660 - core::panicking::panic_fmt::h9440d364e5d01d87
    11:           0x81f5ac - core::panicking::panic::ha6b3ac2de31d736d
    12:           0x4e0ac0 - prjoxide::database::Database::new_builtin::h22a8f7b4adef2479
    13:           0x418b91 - prjoxide::main::h264fe19ae04caa40
    14:           0x409bb6 - std::sys_common::backtrace::__rust_begin_short_backtrace::hd0129080c96afc6c
    15:           0x40a1dc - std::rt::lang_start::{{closure}}::h8d0702a76cfa83f0
    16:           0x78b56f - std::rt::lang_start_internal::hd63699d427712e33
    17:           0x41a708 - main
    18:           0x4013c1 - __tmainCRTStartup
                                 at C:\_\M\mingw-w64-crt-git\src\crt-x86_64-w64-mingw32/C:/_/M/mingw-w64-crt-git/src/mingw-w64/mingw-w64-crt/crt/crtexe.c:333
    19:           0x4014f6 - mainCRTStartup
                                 at C:\_\M\mingw-w64-crt-git\src\crt-x86_64-w64-mingw32/C:/_/M/mingw-w64-crt-git/src/mingw-w64/mingw-w64-crt/crt/crtexe.c:212
    20:     0x7ffdce406fd4 - <unknown>
    21:     0x7ffdcf2fcec1 - <unknown>
C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\MSBuild\Microsoft\VC\v160\Microsoft.CppCommon.targets(238,5): error MSB8066: Custom build for 'C:\Users\User\Desktop\nextpnr\CMakeFiles\9264e4eece416fcd2a7c2b4cbd1744b1\chipdb-LIFCL.bba.rule;C:\Users\User\Desktop\nextpnr\CMakeFiles\c064314d5248ea648493cfbc4f07cfae\chipdb-nexus-bbas.rule' exited wit
h code 101. [C:\Users\User\Desktop\nextpnr\nexus\chipdb-nexus-bbas.vcxproj]
Done Building Project "C:\Users\User\Desktop\nextpnr\nexus\chipdb-nexus-bbas.vcxproj" (default targets) -- FAILED.

Done Building Project "C:\Users\User\Desktop\nextpnr\nexus\chipdb-nexus-bbas.vcxproj.metaproj" (default targets) -- FAILED.

Done Building Project "C:\Users\User\Desktop\nextpnr\chipdb-nexus-bins.vcxproj.metaproj" (default targets) -- FAILED.

Done Building Project "C:\Users\User\Desktop\nextpnr\chipdb-nexus.vcxproj.metaproj" (default targets) -- FAILED.

Done Building Project "C:\Users\User\Desktop\nextpnr\ALL_BUILD.vcxproj.metaproj" (default targets) -- FAILED.

Done Building Project "C:\Users\User\Desktop\nextpnr\nextpnr.sln" (Build target(s)) -- FAILED.

Build FAILED.

"C:\Users\User\Desktop\nextpnr\nextpnr.sln" (Build target) (1) ->
`

Looks like it's trying to find the database.rs file from the source repo, though the Readme implies that should all be included. My commands for building nextpnr looks like:
cmake . -DCMAKE_TOOLCHAIN_FILE="$Env:GITHUB_WORKSPACE/vcpkg/scripts/buildsystems/vcpkg.cmake" -DSTATIC_BUILD=ON -DPython_ADDITIONAL_VERSIONS='3.8' -DCMAKE_CXX_FLAGS='/D "WIN32" /EHsc' -DARCH=nexus -DCMAKE_PREFIX_PATH="$Env:GITHUB_WORKSPACE\prjtrellis" -DOXIDE_INSTALL_PREFIX="$Env:GITHUB_WORKSPACE\prjoxide"

."C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\MSBuild\Current\Bin\MSBuild.exe" /t:Build /p:Configuration=Release /p:Platform=x64 /p:VCTargetsPath="C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\MSBuild\Microsoft\VC\v160\\" nextpnr.sln

The prjoxide just has the bin directory and executable, there's some extra stuff on that cmake command for prjtrellis too of course though the build isn't running for it at the moment.

daveshah1, 2021-01-05:
The database should be being included in the `prjoxide` binary, it shouldn't be being looked for at all. But it's possible if it is missing at **prjoxide build time** something is going wrong - did you do a recursive clone of prjoxide?

davidcorrigan714, 2021-01-05:
Oh that's probably it, should've checked that initially. I'm doing it all in Actions so that I (and others if upstream wants it) will always have up to date nightly windows builds - but it looks like I did forget submodules so I'll rerun and close this out if it fixes them.

davidcorrigan714, 2021-01-05:
Well that fixed that error - got a few build errors with the nexus code in nextpnr, will have to dig into those next.

closed 2021-01-05 by davidcorrigan714