gatecat/prjoxide #9: error with LRAM initial values on LIFCL-17: no IP named LRAM_CORE_R11C0

danc86, 2021-06-10:
I'm targetting LIFCL-17-8UWG72C, trying to use the `INITVAL_00` .. `INITVAL_7F` parameters to fill in initial values for LRAM. prjoxide panics, I guess because it doesn't know the location of all the LRAMs in this part?

```
prjoxide pack lram.fasm lram.bit
thread 'main' panicked at 'no IP named LRAM_CORE_R11C0', prjoxide/src/chip.rs:365:64
stack backtrace:
   0: rust_begin_unwind
             at /rustc/9bc8c42bb2f19e745a63f3445f1ac248fb015e53/library/std/src/panicking.rs:493:5
   1: std::panicking::begin_panic_fmt
             at /rustc/9bc8c42bb2f19e745a63f3445f1ac248fb015e53/library/std/src/panicking.rs:435:5
   2: prjoxide::chip::Chip::get_ip_baseaddr::{{closure}}
   3: prjoxide::chip::Chip::get_ip_baseaddr
   4: prjoxide::chip::Chip::configure_ip
   5: prjoxide::chip::Chip::from_fasm
   6: prjoxide::main
note: Some details are omitted, run with `RUST_BACKTRACE=full` for a verbose backtrace.
```

danc86, 2021-06-10:
I made a "minimal" example, it's not that small because of all the parameters so I will attach the files here (with .txt suffix on the filenames to satisfy Github's rules). I'm not 100% sure the design is valid but Radiant synthesises it at least.
[lram.pdc.txt](https://github.com/gatecat/prjoxide/files/6629376/lram.pdc.txt)
[lram.tcl.txt](https://github.com/gatecat/prjoxide/files/6629378/lram.tcl.txt)
[lram.v.txt](https://github.com/gatecat/prjoxide/files/6629380/lram.v.txt)
I built it by copying the Makefile from another example in the `examples/` subdirectory.

danc86, 2021-06-10:
In case it helps, the resulting .fasm file produced by Yosys seems to refer to the following:

```
IP_LRAM_CORE_R11C0
IP_LRAM_CORE_R15C74
IP_LRAM_CORE_R16C74
IP_LRAM_CORE_R20C0
IP_LRAM_CORE_R2C0
```

gatecat, 2021-06-10:
Can you see if https://github.com/YosysHQ/nextpnr/commit/13c037cc0847d8eea94dabf719ca17dad69254c9 fixes your problem? The basic example seems to build with that, at least.

danc86, 2021-06-11:
Thanks for the fast turnaround! It successfully produces a bitstream for my design now (the design still doesn't work but that's my own mistake I'm pretty sure).

gatecat, 2021-06-11:
Ideally, if you can get it to a point where Radiant works and oxide doesn't then I'll be happy to have a further look for any other issues.

danc86, 2021-06-11:
Thanks for the offer. I think I am at that point now, I have a Litex design which Radiant synthesises and it boots but yosys+nextpnr+prjoxide produces a bitstream that does not boot. I think it must be a separate issue to this one because I am not relying on the LRAM initialisation in the design (just trying to verify that it loads the right values). But it's late on Friday here :-) so next week I will narrow it down and file a separate issue with what I am seeing.

danc86, 2021-06-16:
Okay, seems I was hitting a few issues...

I took out all my LRAM initialization hacks, to make sure the issues were unrelated to this one.

The first problem was that the NX-17 internal oscillator on the board I have runs about 8.6% faster than nominal, which is apparently still within range according to the datasheet. So it turns out in some of my designs, nextpnr was not really meeting timing for the faster clock but it didn't know about it because we declared the period constraint assuming 450MHz. I adjusted the constraint in: https://github.com/google/CFU-Playground/pull/117

But aside from that, there does seem to be a weird problem where the yosys+nextpnr+prjoxide bitstream does not boot but the same design in Radiant works. I filed a new issue for that: #10

danc86, 2021-06-17:
With the fix in #10, I was able to test the LRAM initialisation under yosys+nextpnr+prjoxide and I can confirm this is working nicely -- I have my design executing directly from LRAM :-)

So we can close this issue, unless there is something more you wanted to leave it open for.

closed 2021-06-17 by gatecat