gatecat/prjoxide #19: build prjoxide with new rustc from homebrew package

no111u3, 2021-11-04:
Failed build with rustc 1.56.1 with issue:
```
🍺  /opt/homebrew/Cellar/rust/1.56.1: 30,858 files, 743.1MB
==> Installing twam/openfpga/nextpnr dependency: prjoxide
==> cargo install
Last 15 lines from /Users/nis/Library/Logs/Homebrew/prjoxide/01.cargo:
    | ----------- function or associated item `parse` not found for this
...
207 |     let opts: Opts = Opts::parse();
    |                            ^^^^^ function or associated item not found in `Opts`
    |
    = help: items from traits can only be used if the trait is implemented and in scope
    = note: the following trait defines an item `parse`, perhaps you need to implement it:
            candidate #1: `Parser`

Some errors have detailed explanations: E0432, E0599.
For more information about an error, try `rustc --explain E0432`.
error: failed to compile `prjoxide v0.1.0 (/private/tmp/prjoxide-20211104-70530-1es7s2y/prjoxide-711770c7c7cfc48425827ddba20d13e6dc64b0fe/libprjoxide/prjoxide)`, intermediate artifacts can be found at `/private/tmp/prjoxide-20211104-70530-1es7s2y/prjoxide-711770c7c7cfc48425827ddba20d13e6dc64b0fe/libprjoxide/target`
```

gatecat, 2021-11-05:
Strangely I can't reproduce the issue here, can you try a build again in case there was a transient issue with a dependency or something?

no111u3, 2021-11-05:
I try to clean cache and rebuild only prjoxide and issue still reproduced. Also I build it on M1 mac laptop, but as I can see it isn't architecture depended issues.
```
$ brew install prjoxide
==> Downloading https://github.com/gatecat/prjoxide-db/archive/e31a1b8f214a691d153bbcaa8c3df158d8a22e0b.tar.gz
Already downloaded: /Users/nis/Library/Caches/Homebrew/downloads/b3baa6209f561dc42e03097a933848171c59effd0fee17521c0643e16f8f40bf--prjoxide-db-e31a1b8f214a691d153bbcaa8c3df158d8a22e0b.tar.gz
==> Downloading https://github.com/gatecat/prjoxide/archive/711770c7c7cfc48425827ddba20d13e6dc64b0fe.tar.gz
Already downloaded: /Users/nis/Library/Caches/Homebrew/downloads/d2cf31cc40be08849f31f45e55052f37e0d21d254106a08b60c14826ad73b170--prjoxide-711770c7c7cfc48425827ddba20d13e6dc64b0fe.tar.gz
==> Installing prjoxide from twam/openfpga
==> cargo install
Last 15 lines from /Users/nis/Library/Logs/Homebrew/prjoxide/01.cargo:
    | ----------- function or associated item `parse` not found for this
...
207 |     let opts: Opts = Opts::parse();
    |                            ^^^^^ function or associated item not found in `Opts`
    |
    = help: items from traits can only be used if the trait is implemented and in scope
    = note: the following trait defines an item `parse`, perhaps you need to implement it:
            candidate #1: `Parser`

Some errors have detailed explanations: E0432, E0599.
For more information about an error, try `rustc --explain E0432`.
```

no111u3, 2021-11-07:
Sorry for this issue - fixed by update brew package: https://github.com/twam/homebrew-openfpga/issues/5

closed 2021-11-07 by no111u3