gatecat/prjoxide #24: SEDC support?

femtostar, 2022-05-03:
Lattice Nexus-family FPGAs include SEDC hardware for error detection and correction. A part of the sysCONFIG block, and documented [here](https://www.latticesemi.com/-/media/LatticeSemi/Documents/ApplicationNotes/PT2/FPGA-TN-02076-1-4-SED-SEC-User-Guide-for-Nexus-Platform.ashx?document_id=52788), it exposes a few inputs for configuration and a few outputs for status and error detection information. In Radiant, it can be instantiated from the IP Catalog, which seemingly mostly just generates a wrapper around an instantiated `SEDC` primitive. The instantiation looks like this:

```
SEDC sedc (.SEDENABLE(sed_en_i), 
                .SEDCCOF(sedc_cof_i), 
                .SEDCENABLE(sedc_en_i), 
                .SEDCMODE(sedc_mode_i), 
                .OSCCLKSEDC(oscclk_sedc_w), 
                .RSTSEDC(sedc_rst_w), 
                .SEDCSTART(sedc_start_i), 
                .SEDCBUSY(sedc_busy_o), 
                .SEDCERR(sedc_err_o), 
                .SEDCERRC(sedc_errc_o), 
                .SEDCERRCRC(sedc_errcrc_o), 
                .SEDCERRM(sedc_errm_o), 
                .SEDCFRMERRLOC(sedc_frm_errloc_o), 
                .SEDCDSRERRLOCCIB(sedc_dsr_errloc_o)) ;
```

However, prjoxide does not document this primitive, and it's not included anywhere in the source for prjoxide or nextpnr. Attempting to place and route this with nextpnr-nexus fails with `ERROR: Cell type 'SEDC' instantiated as 'sedc' is not supported by this device.`

It seems like prjoxide just hasn't documented how SEDC works yet. I'm not really familiar enough with the prjoxide fuzzing/development process to try to implement this myself, but it doesn't look like an especially complicated primitive? If I'm wrong and this has already been documented and implemented, how should the SEDC primitive be instantiated when using the yosys+nextpnr+prjoxide toolchain?

gatecat, 2022-05-03:
In theory it's not too difficult to add, although it is indeed currently unsupported. However, for the SEDC to function correctly it would also need the ECCs in the bitstream (which are only used for SEDC and therefore currently don't matter) to be written correctly. From memory there would also need to be some prjtrellis modifications to correctly mask LUTRAM initialization from the ECC calculation which isn't currently done (although most of the ECC logic is correct).

It's a bit of a shameless plug, but if you're using this in a commercial application then it would be possible to get this done quicker if you were willing to take out a support contract with YosysHQ. Otherwise it'd be interesting to know a bit more about your use case for this (you can also contact me privately via the email in my GH profile if you prefer).

femtostar, 2022-05-03:
Well, in terms of our use case, we are the FemtoStar Project - I won't make this a shameless plug for _our_ efforts either, but we're a currently-all-volunteer community of developers, working on a constellation of small satellites for open, privacy-respecting telecommunications. Nothing is orbital yet, but we've made good progress on development, thanks in part to the FOSS FPGA community. Our website is at https://femtostar.com/.

We have been prototyping with the ECP5 and iCE40 (both using yosys+nextpnr, of course - all our work is FOSS, and our toolchains should be too) for over a year, but have recently become interested in the Nexus-family FPGAs, and just yesterday I got some of our code running on an LIFCL-40-EVN. It works, but given our satellites are intended to support operation in orbits of up to 1000 kilometers, radiation is an important consideration, and for that we would like to take advantage of as many reliability features as Nexus has to offer (not just the latch-up-immunity benefits of silicon-on-insulator), so that includes SEDC.

We can, of course, continue to work without it (it's not as if the iCE40 or ECP5 even offer it), but if SEDC is something that could be supported with relative ease, it's certainly something we'd like to have working if we're going to put hardware supporting it in orbit.

gatecat, 2022-05-03:
Sure that seems like an interesting enough use case! I will have a look to see if this can be done.