gatecat/prjoxide #31: incorrect IDCode with LIFCL-40-8BG400CES device

slagernate, 2022-09-14:
Updated the Makefile in examples/blinky/:

````PROJ=blinky
DEVICE=*LIFCL-40-8BG400CES*
PDC=evn.pdc

include ../common.mk```

Ran make but the bitstream (blinky.bit) IDCode is still `0x010F1043` when it should be `0x110F1043`:

```
00000000: 4c53 4343 ff00 00ff ffff bdb3 ffff ffff  LSCC............
00000010: ffff ffff ffff ffff ffff ffff ffff ffff  ................
00000020: 3b00 0000 ffff ffff 3b00 0000 ffff ffff  ;.......;.......
00000030: e200 0000 *010f 1043* 2200 0000 0000 0000  .......C".......
```

Please advise.

gatecat, 2022-09-14:
Are you sure this is wrong? Is it possibly an _ES2_ part you are dealing with, which I think have the same idcode as a production part so you should just remove the ES.

The package and speed grade don't change idcode.

slagernate, 2022-09-14:
Yes, my device is LIFCL-40-8BG400CES2. You are right, my Makefile is not matching my device. I was actually trying to use prjoxide from litex, [where the following devices are allowed](https://github.com/litex-hub/litex-boards/blob/master/litex_boards/targets/lattice_crosslink_nx_evn.py):

```
 target_group.add_argument("--device",        default="LIFCL-40-9BG400C", help="FPGA device (LIFCL-40-9BG400C or LIFCL-40-8BG400CES).")
```
I just assumed 40-8BG400CES was the correct interpretation. I may raise an issue or something in Litex. 
 
Anyway, this was a really annoying hurdle even when just using prjoxide. I think new users would really benefit from just simply including an option for the LIFCL-40-8BG400CES2 device.

slagernate, 2022-09-15:
Also, I don't meant to get off-topic, and likely this is a silly mistake, but I still don't know which device strings are valid because I'm not sure how the `prjoxide --help` command works (i.e. how can I list all the valid devices?). All of the following produced output that didn't make sense to me. 
```
prjoxide --help
prjoxide --device --help
prjoxide device --help
prjoxide -- --device --help
prjoxide -- --device
```

gatecat, 2022-09-15:
Thanks! The best place to deal with this in the stack is in nextpnr, but I've done that in https://github.com/YosysHQ/nextpnr/pull/1025

Your second point is also very valid, so that PR also adds `nextpnr-nexus --list-devices`

closed 2022-09-15 by slagernate