gatecat/prjoxide #38: Unclear error message - High Performance Banks

slagernate, 2023-02-10:
When using e.g. Y5 or W5 (bank 3 / 4 / 5) on the Lattice Crosslink-NX Eval Board, LVCMOS33 doesn't work (not supported). And in my case I was trying to use I2C (BIDIR_LVCMOS33). The error message (below) is helpful in that it lists all available standards, but I think it could benefit from a quick explanation detailing that pins in banks 3, 4 and 5 require the 'H' appended to the standard (AFAIK). Something like "Btw, this bank needs H appended (and only goes up to 1.8v)".

In my (newb) case, I spent 1/2 a day thinking these pins didn't have bidir IO primitives (as even the lattice radiant tool will error for the i2c design I was using) (of course, radiant error messages are completely useless). Only figured it out when I changed the pins from inout to out in the generated litex verilog (I was planning to use open drain instead of bidir IO) and that didn't fix the design. 

Could even go so far as to automatically switch to '*H' standards in the tool (not sure if there are other gotchas here though). 

```
Info: Program finished normally.
thread 'main' panicked at 'No option named BIDIR_LVCMOS33 for enum PIOA.SEIO18.BASE_TYPE in tile CIB_R56C12:SYSIO_B5_0.
Valid options are: BIDIR_HSTL15_I, BIDIR_HSUL12, BIDIR_LVCMOS10H, BIDIR_LVCMOS12H, BIDIR_LVCMOS15H, BIDIR_LVCMOS18H, BIDIR_SSTL135_I, BIDIR_SSTL135_II, BIDIR_SSTL15_I, BIDIR_SSTL15_II, INPUT_HSTL15_I, INPUT_HSUL12, INPUT_LVCMOS10H, INPUT_LVCMOS10R, INPUT_LVCMOS12H, INPUT_LVCMOS15H, INPUT_LVCMOS18H, INPUT_SSTL135_I, INPUT_SSTL135_II, INPUT_SSTL15_I, INPUT_SSTL15_II, NONE, OUTPUT_HSTL15_I, OUTPUT_HSUL12, OUTPUT_LVCMOS10H, OUTPUT_LVCMOS12H, OUTPUT_LVCMOS15H, OUTPUT_LVCMOS18H, OUTPUT_SSTL135_I, OUTPUT_SSTL135_II, OUTPUT_SSTL15_I, OUTPUT_SSTL15_II
Please make sure Oxide and nextpnr are up to date and input source code is meaningful. If they are, consider reporting this as an issue.', prjoxide/src/chip.rs:415:75
```

slagernate, 2023-02-20:
Also, it seems prjoxide won't tell you which pins are causing the issue. E.g. I have a pdc like: 
```
ldc_set_location -site {P1} [get_ports clk]
ldc_set_port -iobuf {IO_TYPE=LVCMOS18H} [get_ports clk]
```

but there's no mention of any signals (e.g. `clk`) in the output:
```
Info: Program finished normally.
prjoxide pack ppm_uart_bridge.fasm ppm_uart_bridge.bit
thread 'main' panicked at 'No option named INPUT_LVCMOS18 for enum PIOA.SEIO18.BASE_TYPE in tile CIB_R56C12:SYSIO_B5_0.
Valid options are: BIDIR_HSTL15_I, BIDIR_HSUL12, BIDIR_LVCMOS10H, BIDIR_LVCMOS12H, BIDIR_LVCMOS15H, BIDIR_LVCMOS18H, BIDIR_SSTL135_I, BIDIR_SSTL135_II, BIDIR_SSTL15_I, BIDIR_SSTL15_II, INPUT_HSTL15_
I, INPUT_HSUL12, INPUT_LVCMOS10H, INPUT_LVCMOS10R, INPUT_LVCMOS12H, INPUT_LVCMOS15H, INPUT_LVCMOS18H, INPUT_SSTL135_I, INPUT_SSTL135_II, INPUT_SSTL15_I, INPUT_SSTL15_II, NONE, OUTPUT_HSTL15_I, OUTPU
T_HSUL12, OUTPUT_LVCMOS10H, OUTPUT_LVCMOS12H, OUTPUT_LVCMOS15H, OUTPUT_LVCMOS18H, OUTPUT_SSTL135_I, OUTPUT_SSTL135_II, OUTPUT_SSTL15_I, OUTPUT_SSTL15_II
Please make sure Oxide and nextpnr are up to date and input source code is meaningful. If they are, consider reporting this as an issue.', prjoxide/src/chip.rs:415:75
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
make: *** [Makefile:34: ppm_uart_bridge.bit] Error 101
```

of course, I can get around this by looking this up in the datasheet / playing around with Radiant / just guessing `LVCMOS18` or `LVCMOS18H` for each signal til I get it right, but clearer error messages would be most ideal :)

gatecat, 2023-02-21:
Agreed, I will assign this to myself to fix in the nextpnr IO handling code.

gatecat, 2023-02-21:
https://github.com/YosysHQ/nextpnr/pull/1105

closed 2023-11-19 by slagernate