gatecat/prjoxide #39: main router loop running indefinitely

slagernate, 2023-02-21:
It seems to never end:
```
Info:     iter=20608 wires=8032 overused=1 overuse=1 archfail=NA
Info:     iter=20609 wires=8032 overused=1 overuse=1 archfail=NA
Info:     iter=20610 wires=8032 overused=1 overuse=1 archfail=NA
Info:     iter=20611 wires=8032 overused=1 overuse=1 archfail=NA
Info:     iter=20612 wires=8032 overused=1 overuse=1 archfail=NA
Info:     iter=20613 wires=8032 overused=1 overuse=1 archfail=NA
Info:     iter=20614 wires=8032 overused=1 overuse=1 archfail=NA
Info:     iter=20615 wires=8032 overused=1 overuse=1 archfail=NA
Info:     iter=20616 wires=8032 overused=1 overuse=1 archfail=NA
Info:     iter=20617 wires=8032 overused=1 overuse=1 archfail=NA
Info:     iter=20618 wires=8032 overused=1 overuse=1 archfail=NA
Info:     iter=20619 wires=8032 overused=1 overuse=1 archfail=NA
Info:     iter=20620 wires=8032 overused=1 overuse=1 archfail=NA
Info:     iter=20621 wires=8032 overused=1 overuse=1 archfail=NA
Info:     iter=20622 wires=8032 overused=1 overuse=1 archfail=NA
Info:     iter=20623 wires=8032 overused=1 overuse=1 archfail=NA
Info:     iter=20624 wires=8032 overused=1 overuse=1 archfail=NA
Info:     iter=20625 wires=8032 overused=1 overuse=1 archfail=NA
Info:     iter=20626 wires=8032 overused=1 overuse=1 archfail=NA
Info:     iter=20627 wires=8032 overused=1 overuse=1 archfail=NA
Info:     iter=20628 wires=8032 overused=1 overuse=1 archfail=NA
Info:     iter=20629 wires=8032 overused=1 overuse=1 archfail=NA
Info:     iter=20630 wires=8032 overused=1 overuse=1 archfail=NA
Info:     iter=20631 wires=8032 overused=1 overuse=1 archfail=NA
^Cmake: *** [Makefile:31: ppm_uart_bridge.fasm] Interrupt
```

I can provide a reproducible example momentarilly...

gatecat, 2023-02-21:
Thanks, this is unfortunately a bit of a long-running issue in nextpnr, new causes seem to keep popping up. A testcase would be appreciated (although, it's worth noting that small changes will usually tip the balance to or from not routing).

slagernate, 2023-04-27:
Seems to happen when using amaranth as well. Can you point me in the right direction if I were to look into nextpnr to fix this? I am curious to not only resolve this, but to have a better understanding of the tools.