gatecat/prjoxide #47: Failed to load chipdb 'nexus/chipdb-LIFCL.bin'

Noxet, 2024-09-24:
I am trying to build a project for a CrossLink-NX device.
The following is being executed by the makefile:

```
pushd /home/noxet/dev/sdi-mipi-video-converter-fpga-design/build/1080p_3g-2lanes && nextpnr-nexus --placer-heap-timingweight 60 -l /home/noxet/dev/sdi-mipi-video-converter-fpga-design/build/1080p_3g-2lanes/1080p_3g-2lanes_nextpnr.log --device LIFCL-40-9BG256C --pdc /home/noxet/dev/sdi-mipi-video-converter-fpga-design/constraints/video_converter_3g-2lanes.pdc --json /home/noxet/dev/sdi-mipi-video-converter-fpga-design/build/1080p_3g-2lanes/1080p_3g-2lanes.json --fasm /home/noxet/dev/sdi-mipi-video-converter-fpga-design/build/1080p_3g-2lanes/1080p_3g-2lanes.fasm && popd
~/dev/sdi-mipi-video-converter-fpga-design/build/1080p_3g-2lanes ~/dev/sdi-mipi-video-converter-fpga-design
ERROR: Failed to load chipdb 'nexus/chipdb-LIFCL.bin'
```

I have built prjoxide from master, and I currently have version 0.1. I cloned the repo recursively.
I also followed the instruction for building nextpnr-nexus.

I do not know how to progress here. Am I missing something?
Thanks

Noxet, 2024-09-26:
Sure.
OS: Ubuntu 22.04

I installed prjoxide:
```
git clone --recursive https://github.com/gatecat/prjoxide
cd libprjoxide
cargo install --path prjoxide
```
 and then nextpnr with: `cmake -DARCH=nexus -DOXIDE_INSTALL_PREFIX=$HOME/.cargo ..` in the build folder.

Then, in the project I am trying to build (an OSS project), when running the makefile I get this output:
```
pushd /home/noxet/dev/sdi-mipi-video-converter-fpga-design/build/1080p_3g-2lanes && nextpnr-nexus --placer-heap-timingweight 60 -l /home/noxet/dev/sdi-mipi-video-converter-fpga-design/build/1080p_3g-2lanes/1080p_3g-2lanes_nextpnr.log --device LIFCL-40-9BG256C --pdc /home/noxet/dev/sdi-mipi-video-converter-fpga-design/constraints/video_converter_3g-2lanes.pdc --json /home/noxet/dev/sdi-mipi-video-converter-fpga-design/build/1080p_3g-2lanes/1080p_3g-2lanes.json --fasm /home/noxet/dev/sdi-mipi-video-converter-fpga-design/build/1080p_3g-2lanes/1080p_3g-2lanes.fasm && popd
~/dev/sdi-mipi-video-converter-fpga-design/build/1080p_3g-2lanes ~/dev/sdi-mipi-video-converter-fpga-design
ERROR: Failed to load chipdb 'nexus/chipdb-LIFCL.bin'
```

gatecat, 2024-09-26:
Have you tried a fully clean build deleting the cmake build folder and trying again?

gatecat, 2024-09-26:
Another thing to check is that nothing went wrong during the build process. Can you confirm that after a build:

 - there's a file `nexus/chipdb-LIFCL.cc` generated inside the build folder 
 
 - the last two lines of that file (it will mostly be a very long blob of data in a string) are:
 ```
 EmbeddedFile chipdb_file_LIFCL("nexus/chipdb-LIFCL.bin", chipdb_blob_LIFCL);
NEXTPNR_NAMESPACE_END
 ```

Noxet, 2024-09-27:
I did a clean build and everything is fine. I realized that I probably forgot to run make install after I got a working build last time... Thank you for your time!